itemisCREATE/statecharts #468: Example of custom code generator

cchesta, 2016-02-26:
Hi, I was wondering if there is some extended example on how to use the Yakindu custom code generator. We would like to use it to automatically generate a make file for C code, would it be possible?

Thank you!
Cristina

stmork, 2016-02-28:
This is an example how to generate a layer between the SCT C++ generator and a Qt application: https://sourceforge.net/projects/sctqtgen/ It should now be easy to generate a makefile.

jdicks, 2016-03-01:
Hi Cristina, 
the sample above might help - thanks @stmork!

Further you can just use the New > Custom Generator wizard to create a simple custom generator. After that you might use the New > Code Generator Model wizard (choose 'Custom Xtend2/Java-based Generator' option on 'Yakindu Generator Model Configuration' page) to create a proper sgen file to simply execute the generated custom generator within your workspace. I expecting you already found https://www.itemis.com/en/yakindu/statechart-tools/documentation/#CreatingCustomCodeGenerators ?!

As you said you want to generate a make file for C code i assume you already generating C code by using the SCT C generator? In this case it will also be possible to extend the shipped C generator so that an additional file would be generated.

Regards, 
Johannes

cchesta, 2016-03-04:
Thanks @stmork and @jdicks: this helps a lot.
Regards,
Cristina

closed 2016-03-15 by andreasmuelder