itemisCREATE/statecharts #475: Inconsistent execution semantics w.r.t. to internal event reactions.

nyssen, 2016-03-02:
When coupling substates of orthogonal regions via raised events, the simulator (and potentially also the code generators) show some inconsistency.

[image]

nyssen, 2016-03-02:
The following models can be used to reproduce the issue.
[default.sct.txt](https://github.com/Yakindu/statecharts/files/154324/default.sct.txt)
[default.sgen.txt](https://github.com/Yakindu/statecharts/files/154325/default.sgen.txt)

jdicks, 2016-03-16:
This came up from several users, see:
* https://groups.google.com/forum/#!msg/yakindu-user/wgkUHokZDZ0/aJt7BJffo7MJ
* https://groups.google.com/forum/?fromgroups#!topic/yakindu-user/bkDvPghMiYk

@terfloth explained related aspects of current solution: 
https://groups.google.com/d/msg/yakindu-user/wgkUHokZDZ0/VOkBzhh_T8EJ

andreasmuelder, 2018-01-02:
Added visual feedback for regino priorities

PR #1835 (2018-01-03): Added decorator for region priority
resolves #475
closed 2018-01-03 by jdicks