itemisCREATE/statecharts #484: [Ctrl+A] does not work

RainerKlute, 2016-03-10:
The user expects to be able to select everything in a text field using [Ctrl+A]. However, this does not work: nothing happens when pressing [Ctrl+A]. (Tested on Linux.)

Please also see issue #430.

jdicks, 2016-05-02:
Ctrl+A does not work for embedded specification editors (xtext editors). properties view and others are working.

The shortcut does not work on windows also.

BeckmaR, 2016-09-13:
This shortcut does work, but not when certain Papyrus plugins are activated.
@RainerKlute please check if this does work in a prebuilt product on linux. Also, you can unselect all Papyrus plugins and readd needed plugins in the launch configuration. This made it work for me.

RainerKlute, 2016-09-14:
As a product, everything works perfectly.

closed 2016-09-14 by RainerKlute

BeckmaR, 2016-09-14:
Maybe we should think about adding this to the testplan. How likely is it that some Papyrus plugin goes rampage and kills this feature in the product - @andreasmuelder?