itemisCREATE/statecharts #491: Eclipse helptext - bad characters

gjsjohnmurray, 2016-03-16:
For example in "Statechart Tools User Guide, SCT, Generating Java source code, The CallHandling example, Creating the statechart model" I see this (notice the two occurrences of â€Š):

Let’s have a closer look at the listing above:
• yakindu::java is the unique ID of the Java code generator. 
•The statechart CallHandling { â€Š } block references the statechart model we want to generate code for. 
•The feature Outlet { â€Š } block specifies where the generated code artifacts are to be placed, i. e. in the Eclipse project CallHandling and within that project in the src-gen folder.

gjsjohnmurray, 2016-03-16:
Just found this already logged in #424

gjsjohnmurray, 2016-03-17:
Under #424 you're asking for clarification about this, so am reopening here with a screenshot attached showing how it affects your help when viewed from within Eclipse, but not at your website.

[image]

RainerKlute, 2016-03-17:
Oh, I see. Thanks for clarification, John!

RainerKlute, 2016-03-22:
Fixed with pull request #514.

closed 2016-03-22 by RainerKlute