itemisCREATE/statecharts #496: SCT simulation triggers workspace build

nyssen, 2016-03-17:
In case of C projects that might lead to execution of make, which is not required for the generation (and make consume a considerable amount of time).

andreasmuelder, 2016-03-22:
@nyssen Is this only the case when the model that is simulated was dirty before starting the simulation and thus an incremental build is triggered when auto-saving the resource?

terfloth, 2016-11-04:
The CDT build won't be triggered on saving a dirty statechart but on starting the simulation. This should not happen. Simulation should work independent of automated build settings.

andreasmuelder, 2016-11-07:
This is the default eclipse behavior, you can disable this via the launch preferences

[image]

Hoewever, we could override this settings to always return false in case of statechart simulation.

andreasmuelder, 2016-11-07:
I opened PR #1067 to never build workspace before launching statechart simulation

closed 2016-11-08 by andreasmuelder