itemisCREATE/statecharts #510: Include wikitext.zip statically for now

RainerKlute, 2016-03-21:
Unfortunately Mylyn WikiText Standalone is available as a nightly build only. Downloading it dynamically during the build is shaky at best, leading to frequent build errors. Until a released WikiText version is available, download the most current wikitext.zip from the link given in http://www.eclipse.org/mylyn/downloads/, include it statically in our repository, and update it as needed.

andreasmuelder, 2016-03-21:
@RainerKlute The link you posted contains a releases update site:
http://download.eclipse.org/mylyn/releases/latest

RainerKlute, 2016-03-21:
@andreasmuelder I need the standalone version.

RainerKlute, 2016-03-21:
Fixed in branch https://github.com/Yakindu/statecharts/tree/issue_510_wikitext.zip.

closed 2016-03-21 by RainerKlute