itemisCREATE/statecharts #528: Decouple JDT dependencies from org.yakindu.sct.commons

jdicks, 2016-03-24:
The JDT dependency was added for accessing classpath entries of a java project during simulation and code generation. this is only needed if java based resources, available in the current runtime workspace, have to be accessed.

To avoid the need of JDT in case no java based resources are necessary these dependencies should be removed in future.

jdicks, 2016-05-12:
plugins dependant from jdt:

org.yakindu.sct.commons
org.yakindu.sct.generator.c
org.yakindu.sct.generator.core
org.yakindu.sct.generator.cpp
org.yakindu.sct.generator.csharp
org.yakindu.sct.generator.csharp.test
org.yakindu.sct.generator.genmodel.ui
org.yakindu.sct.generator.java
org.yakindu.sct.generator.java.test
org.yakindu.sct.simulation.core
org.yakindu.sct.simulation.ui
org.yakindu.sct.standalone

andreasmuelder, 2016-05-13:
This will be fixed via #676

closed 2016-05-13 by andreasmuelder