itemisCREATE/statecharts #536: valueof() throws IllegalArgumentException

andreasmuelder, 2016-03-31:
Declarations:
in event x : integer
var y : integer

State local reaction:
x / y = valueof(x)

If an event with an integer type is defined and raised during simulation, the valueof expression throws an IllegalArgumentException:

Exception in thread "Timer-3" java.lang.IllegalArgumentException: Invalid cast class java.lang.Integer to integer
	at org.yakindu.sct.simulation.core.sexec.interpreter.StextStatementInterpreter._typeCast(StextStatementInterpreter.java:210)
	at org.yakindu.sct.simulation.core.sexec.interpreter.StextStatementInterpreter.typeCast(StextStatementInterpreter.java:666)
	at org.yakindu.sct.simulation.core.sexec.interpreter.StextStatementInterpreter.executeAssignment(StextStatementInterpreter.java:227)
	at org.yakindu.sct.simulation.core.sexec.interpreter.StextStatementInterpreter._execute(StextStatementInterpreter.java:118)
	at org.yakindu.sct.simulation.core.sexec.interpreter.StextStatementInterpreter.execute(StextStatementInterpreter.java:603)
	at org.yakindu.sct.simulation.core.sexec.interpreter.StextStatementInterpreter.evaluateStatement(StextStatementInterpreter.java:108)
	at org.yakindu.sct.simulation.core.sexec.interpreter.DefaultExecutionFlowInterpreter._execute(DefaultExecutionFlowInterpreter.java:355)
	at org.yakindu.sct.simulation.core.sexec.interpreter.DefaultExecutionFlowInterpreter.execute(DefaultExecutionFlowInterpreter.java:588)
	at org.yakindu.sct.simulation.core.sexec.interpreter.DefaultExecutionFlowInterpreter.run(DefaultExecutionFlowInterpreter.java:288)
	at org.yakindu.sct.simulation.core.sexec.interpreter.DefaultExecutionFlowInterpreter.scheduleAndRun(DefaultExecutionFlowInterpreter.java:280)
	at org.yakindu.sct.simulation.core.sexec.interpreter.DefaultExecutionFlowInterpreter.runCycle(DefaultExecutionFlowInterpreter.java:244)
	at org.yakindu.sct.simulation.core.sexec.container.AbstractExecutionFlowSimulationEngine.runCycle(AbstractExecutionFlowSimulationEngine.java:65)
	at org.yakindu.sct.simulation.core.sexec.container.CycleBasedSimulationEngine$1.run(CycleBasedSimulationEngine.java:41)
	at java.util.TimerThread.mainLoop(Unknown Source)
	at java.util.TimerThread.run(Unknown Source)

andreasmuelder, 2016-03-31:
@DominikTesch Please check why the valueof interpreter test didn't fail

terfloth, 2016-03-31:
@andreasmuelder - due to its severity i assign the issue to Johannes. He can fix it today.

andreasmuelder, 2016-03-31:
@terfloth @jdicks  -  already found the problem and I will fix it now. It is a problem with the IntegerEditingSupport, that's why the test model didn't fail.

andreasmuelder, 2016-03-31:
fixed

closed 2016-03-31 by andreasmuelder