itemisCREATE/statecharts #543: Simulation View disappears after terminate

DominikTesch, 2016-04-01:
When you change something in your statechart while running it, and you start a new simulation you must save the new one and you have to terminate the old simulation. Then the new simulation starts without a simulation view.

andreasmuelder, 2016-05-02:
Steps to reproduce:
1. Launch a startchart
2. Change the statechart model, do not save
3. Launch another statechart
4. Save and terminate
Result: Statechart is running with empty Simulation view

andreasmuelder, 2016-05-02:
Fixed via PR ##658 @DominikTesch  Please retest this

DominikTesch, 2016-05-03:
The activ state in the first statechart simulation stays activ after after terminate.

terfloth, 2016-05-09:
@DominikTesch  @andreasmuelder  : isn't the issue reopened due to another problem then the original? Then close this and open a new issue.

andreasmuelder, 2016-05-09:
The merged PR introduces this regression, so reopening this issue is fine.

robert-rudi, 2017-09-19:
could not reproduce initially described issue

closed 2017-09-19 by robert-rudi