itemisCREATE/statecharts #551: Exception when you open a example

DominikTesch, 2016-04-01:
When you open one example, all examples will open (traffic light, digital watch, light switch).
The ftaffic light example throws an NullPointer Exception because of the generator.
Moreover it is the wrong traffic light example

tkutz, 2016-04-01:
@terfloth which traffic light example do we want to ship? There is a new one with swing graphics, but it is currently not included in the product.

tkutz, 2016-04-04:
The problem is related to the order of eclipse builders specified in .project file.
We need the Xtext builder to run first in order to create an index and afterwards run the SCTBuilder to invoke our generator models (.sgen).

terfloth, 2016-04-05:
Changed order of builders on the project.

closed 2016-04-05 by terfloth