itemisCREATE/statecharts #557: Cut-Copy-Paste Keyboard-Shortcuts do not work on OSX

terfloth, 2016-04-04:
Using the context menu actions works as expected.

terfloth, 2016-04-04:
Cmd-X-C-V works in the SCT product. 

In the developer runtime workspace there are additional bindings (when: diagram editing) that seem to conflict with the standard binding (when: in dialogs and windows). It is probably Papyrus stuff...

[image]

tkutz, 2016-06-17:
Yes indeed, it seems to be related to papyrus. Disabling papyrus plugins from launch config helps.

Is there anything to be done for this issue then?

tkutz, 2016-07-15:
@andreasmuelder add reference to bugzilla entry and close the issue

andreasmuelder, 2016-07-26:
This is an external bug by papyrus https://bugs.eclipse.org/bugs/show_bug.cgi?id=481230

closed 2016-07-26 by andreasmuelder