itemisCREATE/statecharts #574: Validation of sematic elements added via the content assist fails

jdicks, 2016-04-04:
If i add a transition and state to the statechart via the semantic element proposals (using strg+space > "add outgoing state"), validation issues will not be shown/will be removed in the graphical editor after saving.

after inserting there should be 2 errors like:
* A state must have a name.	default.sct	/test.184	line: 0 /test/default.sct	Statechart Diagram problems
* Missing trigger. Transition is never taken. Use 'oncycle' or 'always' instead	default.sct	/test.184	line: 0 /test/default.sct	Statechart Diagram problems

The markers are shown in the problems view and other views (e.g. Project Explorer) but never be shown in the SCT editor itself.

When the state or transition is touched (e.g. moved) and the model is saved again the markers are shown as expected.

The problem might be that the sematic element(s) aren't associated properly with a gmf element, so  the marker can't be shown in the editor.

DominikTesch, 2016-04-05:
I add region and an entry point, 2 states (with name) and two transitions via the tools list. If I typ "# x = true" as trigger for the Transition between the states an error mark appears and if I change it to "[x == true]" I will get this errors.
Transitions with an exit point spec does not have a trigger or guard.

When you save the error will disappear but when you change something the error will come back.

jdicks, 2016-07-13:
In another model i observed that 2 types of error markers are shown in problems log.

[image]

jdicks, 2016-10-12:
this includes 2 bugs resulting from
1. transitions properties aren't cleared before the will be parsed again
2. the semantic proposal 'Add outgoing state' didn't add notation elements so that error markers could not be associated to view elements

 see pr #1028

jdicks, 2016-10-19:
was reverted before release 2.8.0 because inserting elements via the semantic proposal leads into overlapping elements. This is caused by absolute bounds set for the notation elements during insert (see PR #1028)

jdicks, 2016-10-19:
extracted #1054
* removing transition properties before parsing again
** solves the inconsistencies mentioned by  DominikTesch (on 5 Apr)
** :exclamation: because it can be applied without further changes

jdicks, 2016-10-19:
To fix inconsistencies mentioned from jdicks on 13 Jul &  on 4 Apr follwoing aspects are important:

* ensure notation elements are serialized even if they are derived during GEF/GMF initialization
** :exclamation: the fix provided in #1028 breaks the automatic calculation of bounds after inserting the mentioned proposal again and leads into overlapping elements in the visual statechart.
** it was observed that currently adding an element via semantic proposal, saving the file, closing and opening it again will lead into lost notation bounds of the inserted element, so that they will be initialized with defaults (-1,-1) and the inserted elements will be shown in the left upper corner
** This is because the notation elements for the added semantic elements, created during diagram initialization will be marked as transient and therefore will not be serialized.
** to fix this and further avoid breaking the calculation of 'non-overlapping' elements after using the semantic proposal more than once it has to be ensured that the notation elements created from a org.eclipse.gef.editpolicies.GraphicalEditPolicy during diagram initialization will be serialized properly.

andreasmuelder, 2017-01-23:
Fixed via enforce persist for new created graphical elements #1171

closed 2017-01-23 by andreasmuelder