itemisCREATE/statecharts #581: Highlighting of taken transitions

tkutz, 2016-04-05:
It seems like the highlighting of taken transitions is not working as expected anymore:

When a transition is taken, it is only briefly highlighted. It would make sense to have a permanent highlighting so that the user understands from which transition a state was entered (see test case T3.2 in test manual)

We should discuss what the expected highlighting strategy is and reflect it in the test manual.

andreasmuelder, 2016-05-02:
@terfloth @tkutz  A taken transition only flashes once, would it be better to highlight the transition as long as the state is active?

terfloth, 2016-05-09:
This could provide a better orientation for the user. We should also consider, that there are more complex scenarios, especially when pseudo states (choice, synch, entry, exit) are involved. We also have to consider high frequency highlighting changes as well as single step debugging modes.

svenjawendler, 2017-05-02:
Now the behavior is as stated by andreasmuelder.

closed 2017-05-02 by svenjawendler