itemisCREATE/statecharts #582: Operation calls without brackets possible

tkutz, 2016-04-05:
Consider a statechart with an operation defined as "operation myOp():boolean". 
This operation can be called as "myOp" instead of "myOp()" in the statechart.

As there is an explicit test case in our test manual (T11.6), I assume this is unexpected behaviour.

To fix this we should:
1. Add a validation
2. Provide better code completion that automatically creates these brackets

tkutz, 2016-04-05:
The possibility to use a function without brackets when it has no input parameters is desired behavior.
The test case is wrong and should be deleted.

RainerKlute, 2016-04-05:
I left the test case in the test plan but specified the correct behavior. Fixed with pull request #590.

closed 2016-04-05 by RainerKlute