itemisCREATE/statecharts #589: Adding a subregion doesn't work if state isn't properly selected

RainerKlute, 2016-04-05:
When selecting a state and pressing `[Ctrl+Space]` a proposal menu appears. One of its options is to add a subregion. However, this doesn't work if the user does not select the state properly, but instead clicks at, say, the local behavior field without editing it. While in such a case `[Ctrl+Space]` still shows the state proposals, nothing happens upon selecting _Add subregion_. BTW, this is different from _Add outgoing state_, which always works.

BeckmaR, 2016-09-13:
Coudln't reproduce, because I can't find a way not to select a state properly. No matter what I do, the thicker border with the 9 black shape modificators always appears.

Does this issue still exist or is this some weird linux editing issue?

RainerKlute, 2016-09-14:
Looks good now! Obviously this was a self-healing bug. :-)

closed 2016-09-14 by RainerKlute