itemisCREATE/statecharts #600: Cannot zoom to selection since selection is deselected

RainerKlute, 2016-04-07:
- Select some part of a statechart you want to zoom to, e.g. one or two states out of many.
- Right-click on the canvas in order to open the context menu and be able to access the _Zoom → Fit to Selection_ menu entry.
- However, upon right-clicking, the selected elements are deselected, and there is not a longer a selection to zoom to.

jdicks, 2016-05-02:
This is as eclipse works in general, whenever a right-click is used the actual mouse position will be passed as context.

RainerKlute, 2016-05-02:
Well, if _Zoom → Fit to Selection_ cannot work when called from the context menu, we shouldn't offer it there, but only in as _Diagram → Zoom → Fit to Selection_ in the main menu.

jdicks, 2016-05-02:
The zoom tools are contributed by the underlying GMF Framework. Please file GMF related bugs into the Eclipse GMF Bugzilla.

closed 2016-05-02 by jdicks