itemisCREATE/statecharts #605: Updatesite suggests SCT 2.6.0 but installs 2.5.0

stmork, 2016-04-09:
See attached screen shots.
[image]
[image]
[image]

tkutz, 2016-04-11:
Thanks for the hint, we will fix this as soon as possible.

For now, please do not check "YAKINDU 2 SDK" which only exists in version 2.5.0 and not for 2.6.0. That should give you a clean installation of the 2.6.0 version.

tkutz, 2016-04-12:
To make it clearer for the user, we added a new category "YAKINDU SCT (old 2.5.0) which only contains 2.5.0 relevant features. The normal YAKINDU SCT category hence does not contain the 2.5.0 SDK feature anymore.

closed 2016-04-12 by tkutz