itemisCREATE/statecharts #610: Changes to sc_types.h are overwritten

stmork, 2016-04-09:
In past versions of SCT the file sc_types.h had a special treatment so that the types can be adjusted to the platform needs. Now the changes are lost after regenerating the C++ artefacts.

nyssen, 2016-04-09:
This can now ne achieved by setting a libraryTargetFolder within the Outlet feature of the sgen file. All static files, i.e. those not depending on the concrete model (including sc_types.h) will be generated into it, and will be preserved upon re-generation. This way, they can now not only be manually adopted but also be shared by different statecharts in one project.

stmork, 2016-04-09:
This simply does not work and the intended behaviour is undocumented.

stmork, 2016-04-09:
I have tried following variants without success:
1. setting `libraryTargetFolder` equal to `targetFolder`
2. setting `libraryTargetFolder` to "."
3. setting `libraryTargetFolder` to a directory different to `targetFolder`

jdicks, 2016-05-23:
pull request #703

jdicks, 2016-05-24:
fixed by https://github.com/Yakindu/statecharts/pull/703

closed 2016-05-24 by jdicks