itemisCREATE/statecharts #611: Standalone CI builds are not documented

stmork, 2016-04-09:
As stated out in your blog post it is now possible to start standalone CI builds as requested in issues #273 and #162. But I cannot find any documentation how to use this.

terfloth, 2016-04-11:
The current solution is preliminary and will be documented by example in a blog post that will be published 12th of April.

jdicks, 2016-04-12:
Hi Steffen,
further info can be found here: https://blogs.itemis.com/embedded/how-to-standalone-generate-statechart-code-for-continuous-integration-builds

closed 2016-04-12 by jdicks

stmork, 2016-10-06:
I've tried the example @jdicks posted on your website (https://blogs.itemis.com/en/how-to-standalone-generate-statechart-code-for-continuous-integration-builds). Unfortunately it does not work with the C++ generator. The reason is that the package of the CPPDefaultFeatureValueProvider is not reexported in the manifest file.

Btw. Is it possible to define the feature libraries using platform URIs like platform:/plugin/...? This seems to be better than copying/extracting the XMIs from the git repository.