itemisCREATE/statecharts #614: Relabel selection tools

RainerKlute, 2016-04-11:
It seems "Marquee Selection" is a term coined by Photoshop, and thus should be replaced by something more common, like "Rectangle Selection". At the same time, find better names and descriptions for these tools in the statechart editor palette. Note that the statechart editor palette comprises two different tools for performing a rectangular selection:

- Only those items are selected that are completely included in the selection rectangle.
- All items are selected that are either completely included in the selection rectangle or touched by it.

For the first tool, the help text should read: "Rectangular inclusion selection: This tool selects items that are completely included in the selection rectangle."

For the second tool, the help text should read: "Rectangular touch selection: This tool selects items that are touched by or included in the selection rectangle."

Both tools' menu entries should be labeled as follows:
- All items
- Nodes only
- Labels only
- Transitions only

DominikTesch, 2016-04-29:
The selection tools are not available in the product. 
Probably your runtime configuration contains a plugin that is not part of the product?

For me it seems like this is coming from GMF, and not from us.

tkutz, 2016-04-29:
@andreasmuelder have the selection tools been removed on purpose or have they never been part of the product / update-site dependencies?

andreasmuelder, 2016-05-02:
These selection tools are added by Papyrus and not part of the product.

closed 2016-05-02 by andreasmuelder