itemisCREATE/statecharts #625: Transition label not shown properly in diagram

RainerKlute, 2016-04-14:
The statechart editor refuses to display the label of one of my transitions. Double-clicking at the transition only produces a small piece of "dirt" on the screen. In the screenshot, the mouse pointer points at it. The transition expression _is_ correctly shown in the properties view, though. This happens on Linux. The model file is attached.

[image]

[LightSwitch.zip](https://github.com/Yakindu/statecharts/files/219236/LightSwitch.zip)

DominikTesch, 2016-04-29:
The model contains the same problem also on Windows
@RainerKlute : can you reproduce it with an any model?

To get out of this, you can simply copy the transition expression from the properties view and create a new transition and paste the expression again.

RainerKlute, 2016-05-02:
I don't even know how to produce or reproduce the error, sorry.

Creating a new transition with the properties of the broken one is a workaround, but no solution. We should look into the model, find out what is wrong, ponder how it might have gotten wrong, and ideally fix the editor to display the "broken" transition correctly.

andreasmuelder, 2018-09-18:
closing as this problem could not be reproduced.

closed 2018-09-18 by andreasmuelder