itemisCREATE/statecharts #634: Simulator fails when using nonintger times for an "after" transition

nfriendly, 2016-04-15:
If you create a transition that involves the keyword after and enter a value that is not a whole number when you attempt to run the statechart a popup occurs that says:
'Launching [StateChartName].sct' has encountered a problem. An internal error occured during: "Launching [StateChartName].sct".

The details dropdown says:
An internal error occurred during: "Launching ServoStateChart.sct".
java.lang.Double cannot be cast to java.lang.Long

Repro:
Create a state chart with two states. Add a transition between the two states and type "after 0.5s".

I couldn't find any documentation saying time values must be whole numbers

andreasmuelder, 2016-04-16:
@nfriendly thanks for reporting

andreasmuelder, 2016-05-02:
The generated Java source code is also broken when using real values for time expressions.
For example, 'after 0.5 s' Is transformed to the expression (0.5 * 1000) which results in a floating type. However, the TimingService API expects an integer for the duration parameter.

andreasmuelder, 2016-05-13:
Add validation rule that only integers are allowed as timing expression

tkutz, 2016-05-31:
When using sub types of integer, the fix does not work.

commit 69bdebd1 (2016-05-27): fixes #634 (#715)

fix #634
closed 2016-06-01 by andreasmuelder