itemisCREATE/statecharts #648: Error generatin state artifacts using .sgen for C code with Yakindu 2.6

MKA-Serenergy, 2016-04-26:
Hello

I have use Yakindu 2.4 with Eclipse Luna for a while now.

Have tried to install Eclipse Mars2 with Yakindu 2.6 and now get issues. When we generate our state chart using "Generate code artifacts" for C code the following error prompts:

_**_An internal error occurred during: "Execute SCT Genmodel APC_CP_SM.sgen".
com.google.inject.internal.util.$ComputationException: java.lang.NoClassDefFoundError: java/util/function/Consumer_**_

It is the same model, which is working in Luna (Yakindu 2.4). we have tried to install Oracle Java version 8 on the Ubuntu 14.04 build machine.This had no effect on issue

andreasmuelder, 2016-05-02:
@MKA-Serenergy  We are not able to reproduce this issue. It seems to be a problem with the Java Version. Can you please paste your eclipse configuration from Help/ About Eclipse/ Installation Details / Configuration. (the properties starting with eclipse.)

MKA-Serenergy, 2016-05-02:
Here you go. I have not included all, since this is very long. If you miss anything, please state category, which I shall copy for you.

I have solved the problem so far, by installing your full 2.6 bundle and building from this environment. Then I use the derived .c and .h file in by own Automotive Eclipse version for complete build.

The next problem is that, the generated code includes unused functions. Your bugfix states that this problem is solved, but it does not seem like it. Therefore we have to disable warnings for the state machine .c file:

set_source_files_properties(statemachine.c PROPERTIES COMPILE_FLAGS -Wno-unused-function)

which is not that great. Are there any build option or other we need to set e.g. in the .sgen file to avoid generating unused function

*** Date: Monday, May 2, 2016 at 11:32:18 AM Central European Summer Time

*** Platform Details:

*** System properties:
applicationXMI=org.eclipse.ui.workbench/LegacyIDE.e4xmi
awt.toolkit=sun.awt.X11.XToolkit
eclipse.application=org.eclipse.ui.ide.workbench
eclipse.buildId=4.4.2.M20150204-1700
eclipse.commands=-os
linux
-ws
gtk
-arch
x86_64
-showsplash
/home/mka/Eclipse Luna Automotive//plugins/org.eclipse.platform_4.4.2.v20150204-1700/splash.bmp
-launcher
/home/mka/Eclipse Luna Automotive/eclipse
-name
Eclipse
--launcher.library
/home/mka/Eclipse Luna Automotive//plugins/org.eclipse.equinox.launcher.gtk.linux.x86_64_1.1.200.v20150204-1316/eclipse_1607.so
-startup
/home/mka/Eclipse Luna Automotive//plugins/org.eclipse.equinox.launcher_1.3.0.v20140415-2008.jar
--launcher.appendVmargs
-exitdata
16000a
-product
org.eclipse.epp.package.automotive.product
-vm
/usr/bin/java
eclipse.home.location=file:/home/mka/Eclipse Luna Automotive/
eclipse.launcher=/home/mka/Eclipse Luna Automotive/eclipse
eclipse.launcher.name=Eclipse
eclipse.p2.data.area=@config.dir/../p2/
eclipse.p2.profile=epp.package.automotive
eclipse.product=org.eclipse.epp.package.automotive.product
eclipse.startTime=1462168328589
eclipse.stateSaveDelayInterval=30000
eclipse.vm=/usr/bin/java
eclipse.vmargs=-Dosgi.requiredJavaVersion=1.7
-XX:MaxPermSize=256m
-Xms40m
-Xmx512m
-jar
/home/mka/Eclipse Luna Automotive//plugins/org.eclipse.equinox.launcher_1.3.0.v20140415-2008.jar
equinox.use.ds=true
file.encoding=UTF-8
file.encoding.pkg=sun.io
file.separator=/
gosh.args=--nointeractive
java.awt.graphicsenv=sun.awt.X11GraphicsEnvironment
java.awt.printerjob=sun.print.PSPrinterJob
java.class.path=/home/mka/Eclipse Luna Automotive//plugins/org.eclipse.equinox.launcher_1.3.0.v20140415-2008.jar
java.class.version=52.0
java.endorsed.dirs=/usr/lib/jvm/java-8-oracle/jre/lib/endorsed
java.ext.dirs=/usr/lib/jvm/java-8-oracle/jre/lib/ext:/usr/java/packages/lib/ext
java.home=/usr/lib/jvm/java-8-oracle/jre
java.io.tmpdir=/tmp
java.library.path=/usr/java/packages/lib/amd64:/usr/lib64:/lib64:/lib:/usr/lib
java.runtime.name=Java(TM) SE Runtime Environment
java.runtime.version=1.8.0_92-b14
java.specification.name=Java Platform API Specification
java.specification.vendor=Oracle Corporation
java.specification.version=1.8
java.vendor=Oracle Corporation
java.vendor.url=http://java.oracle.com/
java.vendor.url.bug=http://bugreport.sun.com/bugreport/
java.version=1.8.0_92
java.vm.info=mixed mode
java.vm.name=Java HotSpot(TM) 64-Bit Server VM
java.vm.specification.name=Java Virtual Machine Specification
java.vm.specification.vendor=Oracle Corporation
java.vm.specification.version=1.8
java.vm.vendor=Oracle Corporation
java.vm.version=25.92-b14
line.separator=

tompiercen, 2016-05-03:
I am observing the same issue. The project works fine with Yakindu 2.4 and 2.5. but fails with 2.6 due to this internal error.
I get the same error with oracle 7 and 8.

jdicks, 2016-05-20:
Hi, 
i was able to reproduce this by explicitly setting the vm via eclipse.ini file to any java 7. As @tompiercen already mentioned, java 8 is currently required during runtime.

The full stacktrace:
```
java.lang.NoClassDefFoundError: java/util/function/Consumer
	at java.lang.Class.getDeclaredConstructors0(Native Method)
	at java.lang.Class.privateGetDeclaredConstructors(Class.java:2493)
	at java.lang.Class.getDeclaredConstructors(Class.java:1901)
	at com.google.inject.spi.InjectionPoint.forConstructorOf(InjectionPoint.java:243)
	at com.google.inject.internal.ConstructorBindingImpl.create(ConstructorBindingImpl.java:96)
	at com.google.inject.internal.InjectorImpl.createUninitializedBinding(InjectorImpl.java:629)
	at com.google.inject.internal.InjectorImpl.createJustInTimeBinding(InjectorImpl.java:845)
	at com.google.inject.internal.InjectorImpl.createJustInTimeBindingRecursive(InjectorImpl.java:772)
	at com.google.inject.internal.InjectorImpl.getJustInTimeBinding(InjectorImpl.java:256)
	at com.google.inject.internal.InjectorImpl.getBindingOrThrow(InjectorImpl.java:205)
	at com.google.inject.internal.InjectorImpl.getInternalFactory(InjectorImpl.java:853)
	at com.google.inject.internal.FactoryProxy.notify(FactoryProxy.java:46)
	at com.google.inject.internal.ProcessedBindingData.runCreationListeners(ProcessedBindingData.java:50)
	at com.google.inject.internal.InternalInjectorCreator.initializeStatically(InternalInjectorCreator.java:133)
	at com.google.inject.internal.InternalInjectorCreator.build(InternalInjectorCreator.java:106)
	at com.google.inject.Guice.createInjector(Guice.java:95)
	at com.google.inject.Guice.createInjector(Guice.java:72)
	at com.google.inject.Guice.createInjector(Guice.java:62)
	at org.yakindu.sct.domain.generic.extension.GenericDomainInjectorProvider.getGeneratorInjector(GenericDomainInjectorProvider.java:159)
	at org.yakindu.sct.generator.core.GeneratorExecutor.getGenerator(GeneratorExecutor.java:86)
	at org.yakindu.sct.generator.core.GeneratorExecutor.executeGenerator(GeneratorExecutor.java:68)
	at org.yakindu.sct.generator.core.GeneratorExecutor$1.run(GeneratorExecutor.java:55)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:55)
Caused by: java.lang.ClassNotFoundException: java.util.function.Consumer
	at java.lang.ClassLoader.findClass(ClassLoader.java:531)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:425)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:358)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:351)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:344)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.loadClass(ModuleClassLoader.java:160)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:358)
	... 23 more
```

jdicks, 2016-05-20:
I figured out that our jenkins executes by using jdk8 - so even if we configure java 7 on PDE & mvn level the xtend compiler generates jdk8 specific code. see [this post](https://groups.google.com/forum/m/#!topic/xtend-lang/anRox6rEGLs) in the xtend-lang user group for further details.

We currently fix this by setting the the execution environment on jenkins explicitly to java 7 but we should solve it by [maven toolchains](http://maven.apache.org/guides/mini/guide-using-toolchains.html) in future.

jdicks, 2016-05-23:
This will be fixed by the upcomming service release. Until the service release is published it is necessary to run SCT within a Java 8 virtual machine.

closed 2016-05-23 by jdicks