itemisCREATE/statecharts #661: "Rename" refactoring doesn't work on Linux

RainerKlute, 2016-05-04:
There are two variants of brokenness of the _Rename_ refactoring on Linux:
- When trying to rename e.g. a variable in the definition section or in the graphical editor, the _Refactoring_ menu appears only very, very briefly, if at all.
- When trying to rename e.g. a variable in the _State Behavior_ section in the _Properties_ view, the selected variable's _Refactoring_ menu appears, but its _Rename …_ entry is inactive.
[image]

jdicks, 2017-05-12:
is related to #1364

RainerKlute, 2017-07-24:
This is working now.

BeckmaR, 2017-10-27:
Cannot confirm, still broken as tested just now and reported by @rherrmannr in #1734.

svenjawendler, 2017-11-14:
With 3.1.1. there is a problem regarding the rename refactoring feature. How to reproduce:
Select an interface within the definition section
Select Refactor - rename in the context menu
In the text field write the new name and click OK
Expectation: every occurence of the interface name will be renamed
Current result: the source name, the interface name is not renamed (< problem), other occurences are

PR #2029 (2018-04-11): Add guards for focus lost with context menu on gtk
Fixes #2023 
Fixes #2022 
Fixes #661
closed 2018-04-11 by andreasmuelder