itemisCREATE/statecharts #662: apply code formatter on generated code

yuecelm, 2016-05-04:
I wrote some code in order to apply the code formatter from my default project "SW core dev" on the generated code. Ideally it should take instead of the hardcoded project the targetProject value from sgen...

yuecelm, 2016-05-04:
Here is the changed code for the cpp generator:

[StatemachineImplementation.patch.txt](https://github.com/Yakindu/statecharts/files/248901/StatemachineImplementation.patch.txt)

[StatemachineHeader.patch.txt](https://github.com/Yakindu/statecharts/files/248902/StatemachineHeader.patch.txt)

yuecelm, 2016-05-04:
Here is the changed code:
[StatemachineHeader.patch.txt](https://github.com/Yakindu/statecharts/files/248908/StatemachineHeader.patch.txt)
[StatemachineImplementation.patch.txt](https://github.com/Yakindu/statecharts/files/248909/StatemachineImplementation.patch.txt)

yuecelm, 2016-05-04:
dont forget to add org.eclipse.cdt.core in Require-Bundle in MANIFEST.MF

andreasmuelder, 2016-05-09:
@yuecelm Thanks you very much for your contribution, applying the CDT formatter to the generated code is a nice idea! 

The problem is, that we don't have a dependency to Eclipse CDT right now in the code generators. Currently, it is possible to generate code without CDT installed, what is really useful if you run in a headless environment for example.

A solution could be to introduce an optional dependency and the formatter is only applied when available. We will schedule this issue for the next release.