itemisCREATE/statecharts #675: Provide a command line interface for generator execution

jdicks, 2016-05-10:
The needed refactorings to create a standalone version of the SCT generators were done for release 2.6.0, see #162.

In the next step there should be a CLI provided to execute Java, C & CPP generators from command line.

This issue includes development of the cli itself & integration in build processes to provide a jar which can be downloaded by users.

andreasmuelder, 2016-08-24:
Done, application will be part of the upcoming 2.8.

closed 2016-08-24 by andreasmuelder

elgeekomm, 2016-11-01:
@andreasmuelder I'm looking for release notes or documentation on this feature, can you confirm this is included in the 2.8 release, and if so can you point to any documentation or instructions? Thanks, this is a HUGELY helpful feature!

andreasmuelder, 2016-11-02:
@elgeekomm In fact, all refactorings required for headless execution are part of 2.8. The application itself is not released as open source since it contains some components of the professional edition. However, it will not require a pro licence and will be free to use. We are currently working on documentation and examples and you can expect this feature to be released within the next 2-3 weeks.

elgeekomm, 2016-11-02:
Excellent news! Thank you for your quick response. This will be a huge help.

andreasmuelder, 2016-11-28:
The headless generator is now available, see  
https://www.itemis.com/en/yakindu/statechart-tools/documentation/user-guide/#generating-state-machine-code-headless
for more details!