itemisCREATE/statecharts #676: Decouple editor from oder SCT features

jdicks, 2016-05-10:
To get the possibility to integrate the SCT editor with less dependencies as possible it should be encapsulated into an own feature. This will lead into the opportunity to ship subsets of features to end users and more robust integration scenarios.

starting point will be the feature 'org.yakindu.sct-feature'

andreasmuelder, 2016-05-12:
Feature dependencies:
org.yakindu.sct.compare-feature -> org.yakindu.sct.editor-feature
org.yakindu.sct.simulation-feature -> org.yakindu.sct.editor-feature
org.yakindu.sct.generator.c-feature -> org.yakindu.sct.generator-feature
org.yakindu.sct.generator.cpp-feature -> org.yakindu.sct.generator-feature
org.yakindu.sct.generator.java-feature -> org.yakindu.sct.generator-feature

andreasmuelder, 2016-05-13:
Check how to keep updatebility

jdicks, 2016-05-20:
To avoid breaking updates the sct.feature will remain in the product for at least one version. It includes the new features so updates will be executed as expected. 
Once the sct.feature is removed updating from older versions without updating to a version including the sct.feature will fail and a clean installation will be necessary.

Added a manual test to our test plan.

closed 2016-05-20 by jdicks