itemisCREATE/statecharts #710: Icon missing in welcome page integration

andreasmuelder, 2016-05-24:
the specification for the content extension does not work in swt mode:

<group style-id="content-group" id="yakindu-introLink-group">
			<link label="YAKINDU Statechart Tools" url="http://statecharts.org"
				id="yakindu-dokumentation-Link" style-id="content-link">
				[image]
				<text>Learn more about YAKINDU Statechart Tools...
				</text>
			</link>
		</group>

The img tag should be replaced by an alt style, see JDT for example.

RainerKlute, 2016-06-20:
Couldn't reproduce the bug: the icon is always there. Neverthelesse I fixed some labels and had the "Tutorials" link indeed point to the tutorials, see pull request #768.

andreasmuelder, 2016-06-23:
@RainerKlute this seems to be platform dependent. The bug is on windows with an installed SCT via update site.

RainerKlute, 2016-06-23:
Sigh! So I'll have to set up a development environment under Windows …

andreasmuelder, 2016-06-23:
We have do provide an alt-style, see for example the JDT
https://github.com/eclipse/eclipse.jdt/blob/master/org.eclipse.jdt/intro/migrateExtensionContent.xml

RainerKlute, 2016-06-23:
Okay, there's something on the issue_#710_welcome branch now that you can test, @andreasmuelder.

RainerKlute, 2016-06-24:
Styling and logo presentation improved.

RainerKlute, 2016-06-27:
Sucessfuly tested on Windows 7, @andreasmuelder.

RainerKlute, 2016-06-28:
I found a way to enter SWT mode. Eclipse must be run with the following properties:
-Dorg.eclipse.swt.browser.DefaultType=mozilla
-Dorg.eclipse.swt.browser.XULRunnerPath=null
This brings up SWT mode on Windows. On Linux, however, the welcome page is completely blank.

RainerKlute, 2016-06-28:
Tested and fine-tuned, see pull request #797.

closed 2016-07-01 by RainerKlute