itemisCREATE/statecharts #712: Document system requirements in short installation guide

tkutz, 2016-05-25:
Besides the comprehensive installation guide, we should have a short overview of the process and dependencies stating the following:

- Current system requirements: JDK 1.8 (and change it to 1.7 once we are again compatible with it)
- Installation process is just unzipping the downloaded file
- When using in Java context: Set preference "Installed JRE" to the installed JDK (this is also required to run our traffic light example)

The current problem is that when SCT is downloaded, the traffic light example does not work out of the box when not setting the JRE preference. This should be documented.

RainerKlute, 2016-05-25:
Won't be worth the effort to document the JDK 1.8 depency, since that's just a bug we are about to fix.

RainerKlute, 2016-06-21:
Seems no JDK is needed at all. JRE is sufficient.

RainerKlute, 2016-06-22:
Can't reproduce the traffic light example problem. The example works for me without having to fiddle around with any JRE preferences.

RainerKlute, 2016-06-22:
Fixed with pull request #773.

RainerKlute, 2016-06-23:
Section on the JRE not found problem added with pull request #776.

closed 2016-06-23 by RainerKlute