itemisCREATE/statecharts #726: Manual event trigger fails for second invocation

tkutz, 2016-06-01:
Consider following statechart:

[image]

When clicking on "e" in simulation view the state change happens correctly. When state A is active again, another click on event "e" does not trigger a state change.

When changing the event value manually by inserting "1" again into the text field, it works fine.

It seems like the label in the simulation view is not updated after an event has been consumed.

tkutz, 2016-07-26:
Fixed with pull request #851.

tkutz, 2016-10-06:
This does not work for events with a type that does not have a default value (e.g. externally defined enumerations).

andreasmuelder, 2016-10-11:
@tkutz is this fixed?

tkutz, 2016-10-11:
This is fixed with pull request #1013

closed 2016-10-11 by tkutz