itemisCREATE/statecharts #735: Statecharts are not scaled correctly with Windows DPI Settings

BeckmaR, 2016-06-03:
When the screen DPI settings in Windows are changed, the text size and the box size in a statechart view is changed, but the position is not. This leads to overlapping states and subcharts and everything. This might be an issue when multiple developers try to work on the same statechart with different screens (I increased the DPI setting because of my 4K screen).

1. Open any statechart with normal DPI settings, check correct look.
2. Save your current work, you'll need to logout later
3. In Win7, right click on desktop, change display settings, go to "make display elements and text larger or smaller" (or similar)
4. Increase DPI setting from 100% to 150%
5. Logout as requested, login, open SCT, open same statechart
6. (Reset DPI setting and logout and login again)

Screenshots attached.

BeckmaR, 2016-06-03:
Note: It looks like the boxes are not scaled at all; some only grow because text is enlarged and needs more lines now.

andreasmuelder, 2018-06-11:
I can not reproduce this with Windows 10. Can someone confirm that this does not happen on Windows 10?

BeckmaR, 2018-06-12:
Looks like Microsoft has fixed something in their graphics libraries then. This would probably mean there isn't much we can do about this behavior on Win7 (Win8?) anyways...

andreasmuelder, 2018-07-05:
This is still an issue on windows 10. The problem is that the Font is adjusted and thus the preferred size of a State is different. The only solution I see here is to avoid font scaling in statechart diagram and use the zoom instead.

This font adjusting must be provided as an option since we do not want to change the layout for existing diagrams modeled with DPI scaling enabled.

andreasmuelder, 2018-11-06:
Add documentation and close

PR #2501 (2018-12-06): updated preference settings for font scaling
fixes #735
commit 0e9dbed9 (2018-12-06): updated preference settings for font scaling (#2501)

fixes #735
closed 2018-12-06 by andreasmuelder