itemisCREATE/statecharts #765: Remove runtime dependencies to generic domain

jdicks, 2016-06-17:
When integrating SCT within a context where an own domain is provided, the generic domain should not be deployed.

If the generic domain is not available there are some Exceptions thrown during runtime. 

it should be possible to:
* run SCT without a generic domain
* open models (*.sct) even if the statecharts domain id could not be found during runtime
  * this might break validations or other functionality, but should not result in exceptions in e.g. SCTResourceFactory or StatechartPaletteProvider

If a configured domain could not be found it might be a good idea to give a feedback to the user.

jdicks, 2016-06-17:
thought about 2 major possibilities:

1. if no default domain could be found and just a single domain is available, use the available domain
2. provide minimal default bindings which could be used in such a case

solution 1 is minimal, but might lead into 'magic' behaviour. solution 2 will need a lot more effort, because providing default bindings might lead into dependencies to concrete implementations which are encapsulated within the generic domain currently.

andreasmuelder, 2016-06-23:
If no default domain is available, there shouldn't be any models in the workspace that have the domainID default so these exceptions are not thrown in that case.

closed 2016-06-24 by jdicks