itemisCREATE/statecharts #799: Package http://www.yakindu.org/base/base/2.0.0 is registered twice

jdicks, 2016-06-29:
It seems the http://www.yakindu.org/base/base/2.0.0 package will be registered twice. It is suspcious that both registrations come from the same source 'org.yakindu.base.types'.

It seems there aren't functional impacts.

On startup the following warning is shown. It is visible in the runtime error log also.
!MESSAGE Both 'org.yakindu.base.types' and 'org.yakindu.base.types' register a package for 'http://www.yakindu.org/base/base/2.0.0'

andreasmuelder, 2016-06-30:
Seems to be that the generated epackage is registered twice in the plugin.xml 
https://github.com/Yakindu/statecharts/blob/master/plugins/org.yakindu.base.types/plugin.xml

andreasmuelder, 2016-07-07:
Fixed with PR #810

closed 2016-07-07 by andreasmuelder