itemisCREATE/statecharts #814: Example wizard: layout problems on Linux

RainerKlute, 2016-07-07:
When the example wizard is started on Linux, it does not layout its dialog window properly. You have to resize the window to trigger the layout.

tkutz, 2016-07-11:
Same for MacOS. Will be fixed with pull request #822.

tkutz, 2016-07-12:
It is fixed for MacOS with pull request #822. 
@RainerKlute: Can you test this on Linux and close this issue if fixed?

RainerKlute, 2016-07-12:
Doesn't work on Linux. Instead of starting, the example wizard throws an exception:
```
!ENTRY org.eclipse.ui 4 0 2016-07-12 10:13:19.143
!MESSAGE Unhandled event loop exception
!STACK 0
java.lang.Error: Unresolved compilation problems: 
	ExampleWizardImages cannot be resolved
	ExampleWizardImages cannot be resolved

	at org.yakindu.sct.examples.wizard.pages.SelectExamplePage.createToolbar(SelectExamplePage.java:102)
	at org.yakindu.sct.examples.wizard.pages.SelectExamplePage.createControl(SelectExamplePage.java:87)
	at org.eclipse.jface.wizard.Wizard.createPageControls(Wizard.java:175)
	at org.eclipse.jface.wizard.WizardDialog.createPageControls(WizardDialog.java:705)
	at org.eclipse.jface.wizard.WizardDialog.setWizard(WizardDialog.java:1153)
	at org.eclipse.jface.wizard.WizardDialog.updateForPage(WizardDialog.java:1211)
	at org.eclipse.jface.wizard.WizardDialog.access$4(WizardDialog.java:1208)
	at org.eclipse.jface.wizard.WizardDialog$8.run(WizardDialog.java:1197)
	…
```

andreasmuelder, 2016-07-12:
@RainerKlute is this fixed now?

RainerKlute, 2016-07-12:
Yes, @andreasmuelder, it is fixed now.

closed 2016-07-12 by RainerKlute