itemisCREATE/statecharts #827: NPE when calling "Generate Code Artifacts"

wendehals, 2016-07-12:
When calling "Generate Code Artifacts" from navigator contect menu an a closed sgen resource, following NPE is thrown:

java.lang.NullPointerException
	at org.yakindu.sct.generator.core.GeneratorExecutor.getGenerator(GeneratorExecutor.java:93)
	at org.yakindu.sct.generator.core.GeneratorExecutor.executeGenerator(GeneratorExecutor.java:68)
	at org.yakindu.sct.generator.core.GeneratorExecutor$1.run(GeneratorExecutor.java:55)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:55)

The sgen file hasn't been opened before.

andreasmuelder, 2016-07-12:
I can only reproduce this when the project the .sgen file is contained in does not have the xtext nature. I
Maybe we should handle this in a more user-friendly way instead of throwing a NPE.

wendehals, 2016-07-14:
Ok, I agree, I only encounter the problem when the project has no xtext nature. Somehow my projetcs were misconfigured. It works now for me.

andreasmuelder, 2016-07-14:
We should add a check in the SCTBuilder if the xtextNature is present. If not we should do nothing.
In the GenerateAction, we should show up the dialog if the user wants to add the XTextNature as in StatechartDiagramEditor#checkXTextNature

terfloth, 2016-09-01:
Derive tasks and close this issue ....

BeckmaR, 2017-06-20:
Closing, followup in #1439

closed 2017-06-20 by BeckmaR