itemisCREATE/statecharts #829: "Link with Editor" does not work

wendehals, 2016-07-12:
The navigator feature "Link with Editor" does not work with sct files.

andreasmuelder, 2016-07-12:
@wendehals I can not reproduce this behavior, it is working as expected for me. Can you provide some more details? Which view does not link? The Project Navigator or Package Explorer? Which version are you using?

wendehals, 2016-07-14:
I'm using SCT 2.6.2. But actually I also can not reproduce the problem anymore. I used the C pespective but it works fine now.

andreasmuelder, 2016-07-14:
OK, I will close this issue then. Please reopen if this happens again.

closed 2016-07-14 by andreasmuelder

wendehals, 2016-07-15:
The problem encounters when the scp file is already opened when eclipse starts and the navigator tree is initially collapsed .