itemisCREATE/statecharts #845: Feature request: generate C header files into separate folder

pkissel, 2016-07-22:
Hello,

First off, thanks for creating SCT.  It is a very useful tool - much easier to use than other statechart tools such as Simulink.

I looked around in the documentation and on the boards and I didn't find any option for splitting the header files generated by the C code generator into a separate directory.  The rationale for this use case is some developers prefer to keep C files and H files in separate folders.  Of course it's easy to copy them but if there was option in the sgen file that would be great.

Thanks

Paul

andreasmuelder, 2016-07-25:
Hi Paul, thanks for your request, this would be a useful feature. 
I schedule this for the upcoming Milestone.

pkissel, 2016-07-25:
Thanks Andreas

tkutz, 2016-08-10:
I added a further outlet parameter "apiTargetFolder" to specify where to generate header files (C / C++ generator) or interfaces (Java generator).

In case "libraryTargetFolder" is specified, headers/interfaces that are state machine model-independent get generated there.

Fixed with pull request #894

closed 2016-08-10 by tkutz