itemisCREATE/statecharts #866: 'Arrange all' throws an exception

jdicks, 2016-08-02:
The following exception is thrown if the tool bar action 'arange all' is used. The action seems to be a default contribution of GMF.

```org.eclipse.core.commands.ExecutionException: While executing the operation, an exception occurred
	at org.eclipse.core.commands.operations.DefaultOperationHistory.execute(DefaultOperationHistory.java:524)
	at org.eclipse.gmf.runtime.diagram.ui.parts.DiagramCommandStack.execute(DiagramCommandStack.java:194)
	at org.eclipse.gmf.runtime.diagram.ui.parts.DiagramCommandStack.execute(DiagramCommandStack.java:169)
	at org.eclipse.gmf.runtime.diagram.ui.actions.DiagramAction.execute(DiagramAction.java:141)
	at org.eclipse.gmf.runtime.diagram.ui.actions.DiagramAction.doRun(DiagramAction.java:103)
	at org.eclipse.gmf.runtime.diagram.ui.actions.internal.ArrangeAction.doRun(ArrangeAction.java:278)
	at org.eclipse.gmf.runtime.common.ui.action.AbstractActionHandler.run(AbstractActionHandler.java:359)
	at org.eclipse.gmf.runtime.common.ui.action.ActionManager$1.run(ActionManager.java:225)
	at org.eclipse.swt.custom.BusyIndicator.showWhile(BusyIndicator.java:70)
	at org.eclipse.gmf.runtime.common.ui.action.ActionManager.run(ActionManager.java:223)
	at org.eclipse.gmf.runtime.common.ui.action.AbstractActionHandler.runWithEvent(AbstractActionHandler.java:377)
	at org.eclipse.ui.actions.RetargetAction.runWithEvent(RetargetAction.java:239)
	at org.eclipse.gmf.runtime.common.ui.action.ActionMenuManager$MenuCreatorAction.runWithEvent(ActionMenuManager.java:216)
	at org.eclipse.jface.action.ActionContributionItem.handleWidgetSelection(ActionContributionItem.java:595)
	at org.eclipse.jface.action.ActionContributionItem.access$2(ActionContributionItem.java:511)
	at org.eclipse.jface.action.ActionContributionItem$6.handleEvent(ActionContributionItem.java:462)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:84)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4362)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1113)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:4180)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3769)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$4.run(PartRenderingEngine.java:1127)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:337)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1018)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:156)
	at org.eclipse.ui.internal.Workbench$5.run(Workbench.java:694)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:337)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:606)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:150)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:139)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:196)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:380)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:235)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:669)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:608)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1515)
	at org.eclipse.equinox.launcher.Main.main(Main.java:1488)
Caused by: java.lang.NullPointerException
	at org.yakindu.sct.ui.editor.policies.EnlargeContainerEditPolicy.getCommand(EnlargeContainerEditPolicy.java:71)
	at org.eclipse.gef.editparts.AbstractEditPart.getCommand(AbstractEditPart.java:500)
	at org.eclipse.gmf.runtime.diagram.ui.editparts.GraphicalEditPart.access$1(GraphicalEditPart.java:1)
	at org.eclipse.gmf.runtime.diagram.ui.editparts.GraphicalEditPart$1.run(GraphicalEditPart.java:482)
	at org.eclipse.emf.transaction.impl.TransactionalEditingDomainImpl.runExclusive(TransactionalEditingDomainImpl.java:328)
	at org.eclipse.gmf.runtime.diagram.ui.editparts.GraphicalEditPart.getCommand(GraphicalEditPart.java:477)
	at org.eclipse.gmf.runtime.diagram.ui.providers.internal.DefaultProvider.createSubCommands(DefaultProvider.java:1149)
	at org.eclipse.gmf.runtime.diagram.ui.providers.internal.DefaultProvider.createNodeChangeBoundCommands(DefaultProvider.java:1117)
	at org.eclipse.gmf.runtime.diagram.ui.providers.TopDownProvider.createNodeChangeBoundCommands(TopDownProvider.java:56)
	at org.eclipse.gmf.runtime.diagram.ui.providers.internal.DefaultProvider.update_diagram(DefaultProvider.java:1051)
	at org.eclipse.gmf.runtime.diagram.ui.providers.internal.DefaultProvider.layoutEditParts(DefaultProvider.java:150)
	at org.eclipse.gmf.runtime.diagram.ui.services.layout.AbstractLayoutEditPartProvider.layoutLayoutNodes(AbstractLayoutEditPartProvider.java:69)
	at org.eclipse.gmf.runtime.diagram.ui.internal.services.layout.LayoutNodesOperation.execute(LayoutNodesOperation.java:74)
	at org.eclipse.gmf.runtime.common.core.service.ExecutionStrategy$1.execute(ExecutionStrategy.java:70)
	at org.eclipse.gmf.runtime.common.core.service.Service.execute(Service.java:651)
	at org.eclipse.gmf.runtime.diagram.ui.services.layout.LayoutService.execute(LayoutService.java:66)
	at org.eclipse.gmf.runtime.diagram.ui.services.layout.LayoutService.layoutLayoutNodes(LayoutService.java:299)
	at org.eclipse.gmf.runtime.diagram.ui.commands.ArrangeCommand.getCommandForExecution(ArrangeCommand.java:124)
	at org.eclipse.gmf.runtime.diagram.ui.commands.ArrangeCommand.doExecuteWithResult(ArrangeCommand.java:93)
	at org.eclipse.gmf.runtime.emf.commands.core.command.AbstractTransactionalCommand.doExecute(AbstractTransactionalCommand.java:247)
	at org.eclipse.emf.workspace.AbstractEMFOperation.execute(AbstractEMFOperation.java:150)
	at org.eclipse.emf.workspace.CompositeEMFOperation.doExecute(CompositeEMFOperation.java:217)
	at org.eclipse.emf.workspace.AbstractEMFOperation.execute(AbstractEMFOperation.java:150)
	at org.eclipse.core.commands.operations.DefaultOperationHistory.execute(DefaultOperationHistory.java:516)
	... 42 more
```

andreasmuelder, 2016-08-08:
Disable arrange toolbar items / context menu from GMF default

tkutz, 2016-08-09:
I cannot reproduce the problem with 2.7.1 product or development runtime.

Using "arrange all" moves the statechart interface box to the right which is not the default setting though. But no exceptions are thrown.

Should we still disable the functionality?

tkutz, 2016-08-10:
Since the functionality does not make any sense in our scenario, I removed the option from toolbar and diagram menu.

Fixed with pull request #889

closed 2016-08-10 by tkutz