itemisCREATE/statecharts #877: java.lang.NullPointerException when exporting using C generator on Pro beta

sylphio, 2016-08-04:
I was simply testing the export with a quick project.  [SCT.zip](https://github.com/Yakindu/statecharts/files/402684/SCT.zip)

I got the following exception in the export log

java.lang.NullPointerException
	at com.yakindu.sct.domain.c.runtime.generator.CTypeSystemAccess.getTargetLanguageName(CTypeSystemAccess.java:20)
	at org.yakindu.sct.generator.c.StatemachineRequiredHeader._functionPrototypes(StatemachineRequiredHeader.java:364)
	at org.yakindu.sct.generator.c.StatemachineRequiredHeader.functionPrototypes(StatemachineRequiredHeader.java:410)
	at org.yakindu.sct.generator.c.StatemachineRequiredHeader._scopeFunctionPrototypes(StatemachineRequiredHeader.java:342)
	at org.yakindu.sct.generator.c.StatemachineRequiredHeader.scopeFunctionPrototypes(StatemachineRequiredHeader.java:399)
	at org.yakindu.sct.generator.c.StatemachineRequiredHeader.content(StatemachineRequiredHeader.java:189)
	at org.yakindu.sct.generator.c.CGenerator.generateArtifacts(CGenerator.java:98)
	at org.yakindu.sct.generator.c.CGenerator.generate(CGenerator.java:89)
	at org.yakindu.sct.generator.c.CCodeGenerator.runGenerator(CCodeGenerator.java:66)
	at org.yakindu.sct.generator.core.impl.AbstractSGraphModelGenerator.doGenerate(AbstractSGraphModelGenerator.java:111)
	at org.yakindu.sct.generator.core.impl.AbstractSGraphModelGenerator.generate(AbstractSGraphModelGenerator.java:81)
	at org.yakindu.sct.generator.core.GeneratorExecutor.executeGenerator(GeneratorExecutor.java:69)
	at org.yakindu.sct.generator.core.GeneratorExecutor$1.run(GeneratorExecutor.java:55)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:55)

sylphio, 2016-08-04:
Look like my operation are causing the error. If i remove them the bug disappear.
operation beginVoltageMonitor()
operation endVoltageMonitor()

tkutz, 2016-08-05:
Importing a working project into Pro version should not produce exceptions. 
I will file an issue in our Pro issue tracker and close this one as it is not open source related.

closed 2016-08-05 by tkutz