itemisCREATE/statecharts #882: Copying subdiagrams creates state with same name, saving not possible

BeckmaR, 2016-08-05:
* create a minimalistic statechart, e.g. two states with transitions
* select both states and refactor->composite
* rename composite state, e.g. LightSwitch
* refactor into subdiagram
* select, copy and paste the state containing the subdiagram

Now, a second state containing the subdiagram is created, that has the same name. **No error marker** is shown. When the subdiagram is edited, SCT says is failed to save the statechart, due to something not being contained in a container. Ignoring this it can be seen that the subdiagram of both states was changed (same reference or something -> the subdiagram is not properly copied).

Might be related to #802 and #803.

BeckmaR, 2016-08-05:
Completely solved with today's pull request. Closing.

tkutz, 2016-08-11:
I can observe the same problem when the composite state is inside another composite state.

See example below. Copying A creates another A state and saving results in error.
[image]

tkutz, 2016-08-17:
The problem occurs non deterministically. GMF stores the copy content in a set, which gets ordered by IDs, which are different from diagram to diagram. In case the subdiagram is to be copied before the node, a copyAlways link from diagram to its element reference (here state A) is created. GMF then tries to copy state A after it has copied the diagram, but this fails due to several reasons among one is that our own DiagramPasteOperation has the resource as paste target which is different to proper paste target of the state.

To fix this, I prohibit the creation of the copyAlways link from a diagram its referenced state. Then the link is created between the node to be copying the state A. This works fine.

tkutz, 2016-08-18:
Fixed with pull request #907.

closed 2016-08-18 by tkutz