itemisCREATE/statecharts #890: Example wizard: initial layout

jdicks, 2016-08-09:
If starting the example wizard it is hard to get an overview of the examples without collapsing all the nodes in visible tree, find screenshot below.

The following things might improve the current situation, see also 2nd screenshot
* collapse all top level nodes
* select an example
* enlarge dialog dimension
* there might be a better order than an alphabetical (basic->advanced)

Current:
[image]

Draft initial layout:
[image]

terfloth, 2016-08-30:
This tasks is mainly about the usability of the example wizard. We should derive different tasks from this issue:

- implement basic improvements like collapsing top level nodes
- Apply a usability tests to this feature.

andreasmuelder, 2016-11-18:
Not required since we provide the examples as flat list

closed 2016-11-18 by andreasmuelder