itemisCREATE/statecharts #891: Example Wizard: Add option to sort by name

tkutz, 2016-08-10:
Currently, examples are sort by category, but since an example has multiple categories, they all appear multiple times. This is confusing.

Solution A:
* Provide radio button to switch between "sort by category" and "sort by name". The second option would then just show all the examples sorted alphabetically

Solution B:
* Provide checkbox "Show categories". If enabled, we get the current behavior, if disabled, examples are only shown with their names and not categorized

terfloth, 2016-08-30:
As there are other options for handling n:m relation between categories and examples and as it is basically a usability issue we should derive UX tasks.

andreasmuelder, 2016-11-18:
for the next release we will show a simple flat list. This is sufficient for ~8 example projects.

closed 2016-11-22 by andreasmuelder