itemisCREATE/statecharts #912: Design smell in IGenArtifactConfigurations API

wendehals, 2016-08-18:
The interface org.yakindu.sct.generator.c.IGenArtifactConfigurations there is a method

List<GenArtifactConfiguration> getConfigurations();

that has a design smell. The type parameter GenArtifactConfiguration of the returned list is actually an inner class of org.yakindu.sct.generator.c.GenArtifactConfigurations that implements IGenArtifactConfigurations. Thus, there is no real decoupling between the interface and its implementation.

Consider to create an interface IGenArtifactConfiguration for the type parameter of the getConfigurations() method.

tkutz, 2016-08-19:
Ok I see, will keep that in mind when refactoring the generators anyways.

andreasmuelder, 2016-08-24:
done with PR #921

closed 2016-08-24 by andreasmuelder