itemisCREATE/statecharts #918: Documentation regarding transit sequence is wrong

terfloth, 2016-08-24:
The documentation states:

9.2.4 Events specification:
      "entry - State has just become active."

9.2.3.1 Fundamental statechart methods:
       "3. If a state change has to be done:
                        1. Execute exit actions of the present state.
                        2. Save history state, if necessary.
                        3. Make the new state active.
                        4. Execute entry actions of the new state."

The actual sequence for a transition from A -> B is

1. mark A inactive
2. perform exit actions of A
3. perform transition actions
4. perform entry actions of B
5. mark B active

RainerKlute, 2016-08-24:
Where's saving the history state in your sequence, @terfloth?

RainerKlute, 2016-08-24:
Fixed with pull request #920.

closed 2016-09-01 by andreasmuelder