itemisCREATE/statecharts #919: Consider creating some useful SText templates

RainerKlute, 2016-08-24:
Which templates would be helpful to the user? Random thoughts:
- Transitions specification with event trigger, guard condition, effect
- Sample state behaviour
- Sample variable, event, etc. declarations
- Large, full-blown templates serving as examples, but might produce errors due to e.g. a variable assignment
- Small, error-free templates

svenjawendler, 2017-06-26:
One participant of the usability dinner Dortmund/2017 had a problem related to this: he was not really able to create an event.

svenjawendler, 2018-02-07:
1 Transitions specification with event trigger, guard condition, effect
2 Sample state behaviour
3 Sample variable, event, etc. declarations
 - int variable
 - const
 - readonly real variable (pi)
 - in event
 - out event
 - example operation
4 Large, full-blown templates serving as examples, but might produce errors due to e.g. a variable assignment
5 Small, error-free templates

svenjawendler, 2018-02-26:
There are templates.

closed 2018-02-26 by svenjawendler