itemisCREATE/statecharts #948: Weird display when editing interface text box in neon

mistoll, 2016-09-06:
System:
ubuntu 14.04.5 desktop (both x64 and x86)
Java SE Runtime 1.8.0_101-b13
Eclipse Neon 20160613-1800

When entering the interface text box in order to edit it the display jumps from 

> interface

to 

> i
>
> n
> 
> t
> 
> e
> 
> r
> 
> f
> 
> a
> 
> c
> 
> e
> 

Then any mouse interaction seams broken. Text input is handled correctly. While any input is displayed one character per two lines, the box shows the correct content when loosing focus.

RainerKlute, 2016-09-07:
@mistoll, what exactly do you mean by "interface text box"?

mistoll, 2016-09-07:
According to the dokumentation the box is called "definition section" (of the statechart editor).

RainerKlute, 2016-09-09:
Ah, thanks. I could reproduce this bug, and it seems to occur only if you use Gtk3 and Neon. While we are planning a workaround for the next release, *your* workaround for now could be to either use Mars instead of Neon or to use Gtk2 instead of Gtk3. In order to do the latter, start Eclipse with the following command:
```SWT_GTK3=0 eclipse```
Hope this helps for the time being!

mistoll, 2016-09-09:
I can confirm, with SWT_GTK3=0, the definition section is displayed correctly.

BeckmaR, 2016-10-18:
As this seems to be one of the bugs related to SWT and GTK3, we can't fix it. The workaround can be found in our [wiki](../wiki/Known-Linux-issues-and-workarounds).

closed 2016-10-18 by BeckmaR