itemisCREATE/statecharts #960: [yakindu pro] After the online update, Yakindu pro cannot load my chart file.

sylphio, 2016-09-07:
Guice creation errors:

1) Error injecting constructor, java.lang.ClassCastException: org.eclipse.xtext.impl.GroupImpl cannot be cast to org.eclipse.xtext.Alternatives
  at org.yakindu.base.expressions.services.ExpressionsGrammarAccess.<init>(Unknown Source)
  at org.yakindu.base.expressions.services.ExpressionsGrammarAccess.class(Unknown Source)
  while locating org.yakindu.base.expressions.services.ExpressionsGrammarAccess
    for parameter 1 at org.yakindu.sct.model.stext.services.STextGrammarAccess.<init>(Unknown Source)
  at org.yakindu.sct.model.stext.services.STextGrammarAccess.class(Unknown Source)
  while locating org.yakindu.sct.model.stext.services.STextGrammarAccess
    for field at org.yakindu.sct.model.stext.validation.STextJavaValidator.grammarAccess(Unknown Source)
  while locating com.yakindu.sct.domain.c.runtime.validator.CSTextJavaValidator
  at org.eclipse.xtext.service.MethodBasedModule.configure(MethodBasedModule.java:57)
  while locating org.yakindu.sct.model.stext.validation.STextJavaValidator

1 error
Guice creation errors:

1) Error injecting constructor, java.lang.ClassCastException: org.eclipse.xtext.impl.GroupImpl cannot be cast to org.eclipse.xtext.Alternatives
  at org.yakindu.base.expressions.services.ExpressionsGrammarAccess.<init>(Unknown Source)
  at org.yakindu.base.expressions.services.ExpressionsGrammarAccess.class(Unknown Source)
  while locating org.yakindu.base.expressions.services.ExpressionsGrammarAccess
    for parameter 1 at org.yakindu.sct.model.stext.services.STextGrammarAccess.<init>(Unknown Source)
  at org.yakindu.sct.model.stext.services.STextGrammarAccess.class(Unknown Source)
  while locating org.yakindu.sct.model.stext.services.STextGrammarAccess
    for field at org.yakindu.sct.model.stext.validation.STextJavaValidator.grammarAccess(Unknown Source)
  while locating com.yakindu.sct.domain.c.runtime.validator.CSTextJavaValidator
  at org.eclipse.xtext.service.MethodBasedModule.configure(MethodBasedModule.java:57)
  while locating org.yakindu.sct.model.stext.validation.STextJavaValidator

1 error

Hi, 

[SCT.zip](https://github.com/Yakindu/statecharts/files/460192/SCT.zip)
I have attached the problematic files.

Thank
Seb

andreasmuelder, 2016-09-08:
Hi Seb, 
thanks for reporting. We will disable the update feature as long as SCT pro is still beta to avoid such problems. The available beta of the pro edition is not intended to be updated with the current available version 2.7.1 of the open source edition.  We will release an updated version of SCT pro on 23th of September. Until then you should use the bundled pro version without updating. 
Best,
Andreas

sylphio, 2016-09-08:
hi, 
   Good to know,  I did not realize the eclipse update was mixing both version.
Thank for the answer
Seb

P.s I use Yakindu pro for a personal project and it is really good. Thank.

closed 2016-09-08 by sylphio