itemisCREATE/statecharts #979: Error in statechart language parser

RainerKlute, 2016-09-16:
Within a state, I use local reactions like the following:
```
[guard1] / action1
[guard2] / action2
```
This works as expected. However, now I change that to the following:
```
[guard1] / raise event
[guard2] / action2
```
The parser now complains about "missing EOF at '/'" at the "/" in the seconds line.

And here's a work around:
```
[guard1] / raise event
always [guard2] / action2
```

RainerKlute, 2016-09-20:
As discussed with @andreasmuelder, this error is a consequence of the upcoming array support. The syntax for accessing array elements should be changed, or otherwise existing models would be broken.

BeckmaR, 2016-09-20:
How about the Xtend-ey array.get( _index_ )