itemisCREATE/statecharts #1003: Could not execute internal transition???

hongsonitptit, 2016-10-03:
Hello, 
I am trying to build a simple statechart as image bellow:
[image]
It worked perfectly in simulation mode, but after I genarated it to C++ or JAVA. It has a problem!!!
As you see in the my image, one transition will be executed automatically when lightA == true & lightB == true.
It worked automatically when simulating time, but when I ran code generated by it, if I call event as scenario:
1. call raiseOnA()
2. call runCycle() --> state is OFF, light A = true
3. call rasiseOnB()
4. call runCycle() --> state still OFF, ligth B = true
5. call runCycle() again --> execute doSomethin() --> state is ON, light A = true, light B = true

My problem is I do not understand why I need call runCycle() 2 times to go state ON, why it do not work automatically??
If I have many sequence self-transition (do not know before), I think it will be inconvenience when programming. 
If you can explain that, or may be I am misunderstand something in your tool?

BeckmaR, 2016-10-04:
Hello,

the behavior you are seeing is completely correct the way it is. In every run-cycle, transitions are evaluated before the state reactions. This means that after your local reaction `onB / lightB = true` is executed, the transition won't be evaluated once more, and the transition is only taken in the next run-cycle. Everything else would lead to faulty behavior - I would not want my local reactions to be executed when the state should be left in this run-cycle. Imagine a counter that counts every cycle inside of a state, it should not count the leaving cycle normally. So yeah, you have to call runcycle twice. The simulation calls runcycle every 200ms per default, so that's why you are not seeing it in the simulation. The behavior should still be the exact same and this is not a bug.

Regards,
René

closed 2016-10-04 by BeckmaR

hongsonitptit, 2016-10-04:
Thank for your feedback
I see your way in RTC rule, but I hope you update project because if you have many sequence transitions, your state chart tool will work more perfectly.
If you do not want to execute many transition (or something like a loop), I think you can allow user to set up maximum transition time for their state chart.