itemisCREATE/statecharts #1005: "Install from update site" not possible after registering

BeckmaR, 2016-10-04:
This button ->
[image]
does nothing, it only reloads the site with a new link. Tested on linux with firefox and chrome.

andreasmuelder, 2016-10-10:
@RainerKlute can you reproduce this on Linux?

RainerKlute, 2016-10-11:
Confirmed. Tested with current versions of Firefox, Chromium, and Konqueror.

(It also turned out that the registration form on https://info.itemis.com/download-yakindu-statechart-tools is not displayed when the Privacy Badger https://www.eff.org/de/privacybadger is used. The Privacy Badger detects six trackers on that webpage and deactivates the five Hubspot trackers.)

andreasmuelder, 2016-11-10:
Reported as fixed from our webmaster

closed 2016-11-10 by andreasmuelder